akafael/ufpe-project-bsg #94: Update DOXYFILE to include pending tools

akafael, 2023-09-21:
Some tools are missing at

https://github.com/akafael/ufpe-project-bsg/blob/31731380427c6d36e553e9b3442edabbc67fa8a9/DOCKERFILE#L4

Add the step to install and configure the following tools

* [ ] Doxygen
* [ ] lcov
* [ ] gcovr
* [ ] pre-commit

Some steps are described at 

LCOV + GCOVR
https://github.com/akafael/ufpe-project-bsg/blob/31731380427c6d36e553e9b3442edabbc67fa8a9/.github/workflows/c-cpp.yml#L26

Doxygen
https://github.com/akafael/ufpe-project-bsg/blob/31731380427c6d36e553e9b3442edabbc67fa8a9/.github/workflows/c-cpp.yml#L38

pre-commit
https://github.com/akafael/ufpe-project-bsg/blob/31731380427c6d36e553e9b3442edabbc67fa8a9/.github/workflows/c-cpp.yml#L15

akafael, 2023-09-21:
Suggestion: add the following lines to the dockerfile

```dockerfile
RUN apt-get install gcc lcov gcovr
RUN apt-get install doxygen graphviz
RUN pip install pre-commit
```

SSDaniel, 2023-09-21:
Done.

closed 2023-09-21 by akafael