akafael/ufpe-project-bsg #120: Explain bsg.c differencea between them and bsgcontrol.c

SSDaniel, 2023-09-28:


akafael, 2023-09-28:
`bsbcontrol` represents the core system functionality. All routine related to controlling the actual BSG device should be placed there.

`bsg.h` is just a place to store the struct BSG used as [Interface](https://en.wikipedia.org/wiki/Interface_(object-oriented_programming)) that represents a pack of information related to the BSG sensors and possible commands. 

`bsg.c` was created to store helper functions related to the struct BSG. For instance, a function to compare if two structs are equal or a helper function that receives the struct BSG and translates into real command to a real hardware device.

SSDaniel, 2023-10-03:
That's Enough.

closed 2023-10-03 by SSDaniel